Freetz/freetz #206: ltrace: error (due to used compiler options) in architecture dependent code, while compiling for ATOM processor with speed optimizations

PeterPawn, 2019-06-29:
compressed configuration file to reproduce (attached as file, too):
[config.compressed.txt](https://github.com/Freetz/freetz/files/3341656/config.compressed.txt)

```
FREETZ_USER_LEVEL_EXPERT=y
FREETZ_TYPE_6490=y
FREETZ_TYPE_FIRMWARE_07_1X=y
FREETZ_PACKAGE_LTRACE=y
FREETZ_TARGET_CFLAGS="-Ofast -pipe"
```

output (reduced to the error) from build of `ltrace` package (*complete* log file is attached, too):
[fmake.txt](https://github.com/Freetz/freetz/files/3341662/fmake.txt)

```
libtool: compile:  /home/freetz/atom_ltrace/toolchain/build/i686_gcc-5.5.0_uClibc-1.0.14-nptl/i686-linux-uclibc/bin/i686-linux-uclibc-gcc -DHAVE_CONFIG_H -I. -I../../.. -DSYSCONFDIR=\"/etc\" -DPKGDATADIR=\"/usr/share/ltrace\" -I/home/freetz/atom_ltrace/toolchain/build/i686_gcc-5.5.0_uClibc-1.0.14-nptl/i686-linux-uclibc/usr/include -I../../../sysdeps/linux-gnu/x86 -I../../../sysdeps/linux-gnu -I../../../sysdeps -I../../.. -Wall -Wsign-compare -Wfloat-equal -Wformat-security -Werror -march=atom -mtune=atom -Ofast -pipe -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -MT fetch.lo -MD -MP -MF .deps/fetch.Tpo -c fetch.c  -fPIC -DPIC -o .libs/fetch.o
fetch.c: In function 'classify_eightbyte':
fetch.c:377:33: error: 'cls2' may be used uninitialized in this function [-Werror=maybe-uninitialized]
   else if (cls == CLASS_INTEGER || cls2 == CLASS_INTEGER)
                                 ^
cc1: all warnings being treated as errors
make[5]: *** [Makefile:435: fetch.lo] Error 1
```

The problem doesn't occur with Freetz's default compiler options for optimization (-Os) - only if speed optimization (-Ofast) is requested (expert level in ".config" is only needed to allow modification of CFLAGS settings).

er13, 2019-06-29:
Ich würde sagen, der Compiler hat Recht.

[cls2](https://github.com/dkogan/ltrace/blob/82c66409c7a93ca6ad2e4563ef030dfb7e6df4d4/sysdeps/linux-gnu/x86/fetch.c#L363) wird initial kein Wert zugewiesen. Die einzige Stelle, an der cls2 überhaupt ein Wert zugewiesen wird, ist der Aufruf von [classify](https://github.com/dkogan/ltrace/blob/82c66409c7a93ca6ad2e4563ef030dfb7e6df4d4/sysdeps/linux-gnu/x86/fetch.c#L368). Und der Ablauf von [classify](https://github.com/dkogan/ltrace/blob/82c66409c7a93ca6ad2e4563ef030dfb7e6df4d4/sysdeps/linux-gnu/x86/fetch.c#L456) ist nun mal so, dass diese [verlassen werden kann](https://github.com/dkogan/ltrace/blob/82c66409c7a93ca6ad2e4563ef030dfb7e6df4d4/sysdeps/linux-gnu/x86/fetch.c#L464), ohne dass `classes[]` (entspricht `&cls2`) ein Wert zugewiesen wird.

Vom Lesen her würde ich sagen für `void` müsste der Wert `CLASS_NO` lauten. `CLASS_NO` taucht an insgesamt 5 Stellen auf. Interessant ist [diese](https://github.com/dkogan/ltrace/blob/82c66409c7a93ca6ad2e4563ef030dfb7e6df4d4/sysdeps/linux-gnu/x86/fetch.c#L373) - `cls2` wird auf `CLASS_NO` getestet, obwohl sie diesen Wert gar nicht haben kann - denn die einzige Stelle, wo `CLASS_NO` auf der rechten Seite einer Zuweisung auftaucht ist [diese](https://github.com/dkogan/ltrace/blob/82c66409c7a93ca6ad2e4563ef030dfb7e6df4d4/sysdeps/linux-gnu/x86/fetch.c#L361).

Daher halte ich den eingecheckten Fix für wahrscheinlich richtig ;-) Es sei aber trotzdem darauf hingewiesen, dass dieser nur compile-tested ist.

PeterPawn, 2019-06-29:
OK, ich hatte mich für den einfacheren Weg (nach meiner Ansicht) entschieden (https://github.com/PeterPawn/YourFreetz/commit/0b1ceba8e6e624d47b60102c234bc37d277553dd) und `cls2` einfach einen definierten Wert bei der Initialisierung (bzw. am Beginn der "lifetime") zugewiesen - und zwar auch den für die einfache Anzeige als void-Pointer ("Standard" wäre ja "CLASS_INTEGER", wenn der enum-Wert auf 0 initialisiert wird vom Compiler).

Meine Philosophie dahinter: Alles ist erst mal "void" bei den Argumenten und Rückgabewerten, solange es keinem genauer spezifizierten Datentyp (mit speziellerer Darstellung und eigenem Wertebereich) zugeordnet werden kann.

Solange dieser vorbelegte Wert nicht innerhalb von `classify()` geändert wird, bleibt es bei dem aus der Initialisierung - im anderen Falle gilt die Behauptung

> may be used uninitialized

ja auch nicht länger, wenn die Variable dann doch mit einem anderen Wert versehen wird (egal in welcher Schachtelungstiefe).

Wichtig ist nur (so, wie ich es interpretiere), daß [an dieser Stelle](https://github.com/dkogan/ltrace/blob/82c66409c7a93ca6ad2e4563ef030dfb7e6df4d4/sysdeps/linux-gnu/x86/fetch.c#L371) dann entweder ein definierter Wert in `cls2` steht oder von `classify()` etwas anderes als `CLASS_NO` aus der Initialisierung gesetzt wurde, damit nicht die Daten am Ende noch als Gleitkommazahlen interpretiert werden (SSE und x87 wird da wohl dann zusammengefaßt). Ohne die Initialisierung würden die Daten dann wohl erneut als CLASS_INTEGER klassifiziert werden (wie *fast* alles andere, wenn man in `classify()` nachsieht) - vermutlich juckt(e) es deshalb bisher niemanden in den vergangenen 5 1/2 Jahren, seitdem es so in `ltrace` steht (und ohne entsprechende Compiler-Optionen bricht die Übersetzung hier ja auch nicht unbedingt ab).

--

Beide Patches sollten aber das Problem lösen, (wenn der Compiler tatsächlich bis in den Control-Flow in `classify()` nach einer Zuweisung sucht - wenn Du "compile-tested" schreibst, wird die Meldung bei Dir ja ebenfalls "stumm" sein nach dem Patch. 

Ich finde die Variante, das "an Ort und Stelle" zu initialisieren und nicht erst nach dem bzw. im Rahmen des Funktionsaufrufs, halt übersichtlicher und leichter zu erkennen beim Review. Zumal wieder `classify_eightbyte()` (über `classify_eightbytes()`) rekursiv aus `classify()` heraus aufgerufen werden kann, wenn es sich nicht um "simple data types" handelt - da kommt vielleicht der Flow-Analyzer des Compilers noch mit, aber für den "menschlichen Leser" wird's deutlich schwerer.

--

Wer macht hier zu? Ich wollte das meinerseits ohnehin "nur melden" ... schon die Umschaltung auf Speed-Optimierung beim Kompilieren ist sicherlich kein *Standardverhalten* eines Freetz-Users und muß damit wohl auch nicht unbedingt unterstützt werden (der zweite Grund, warum ich keinen PR aufgemacht habe - ich gehe/ging trotzdem davon aus, daß Du den Branch gemeldet bekommen hast vom GitHub). 

Funktioniert Dein Patch auch bei mir, verwerfe ich meinen Branch `ltrace` beim nächsten `rebase` automatisch wieder.

er13, 2019-07-02:
> Ich finde die Variante, das "an Ort und Stelle" zu initialisieren und nicht erst nach dem bzw. im Rahmen des Funktionsaufrufs, halt übersichtlicher und leichter zu erkennen beim Review. ... da kommt vielleicht der Flow-Analyzer des Compilers noch mit, aber für den "menschlichen Leser" wird's deutlich schwerer.

Ich habe das für C typische Pattern gesehen
```
type variable;
returnCode = some_function(..., &variable, ...);
if (returnCode < 0)
    return error;

code assuming variable is properly initialized
```
und deswegen war es für mich natürlicher eben nach den Fehlern in `some_function` zu suchen. Und da fand sich schnell eben eine Stelle, wo einerseits `success` zurückgegeben wird und andererseits `variable` jedoch uninitialized blieb.

Am Ende, im Kontext von `classify_eightbyte`, ist es aber egal - beide Patches haben dieselbe Wirkung: ob man zu Beginn der Life-Time initialisiert und dann in einem einzigen Fall den Wert behält und in den anderen überschreibt oder eben diesen einzigen Fall um den vergessenen (2.ten) Rückgabewert vervollständigt, ist dasselbe in Grün.

Ansonsten überrascht mich, dass es die Warning nur bei `-Ofast` gibt. Meine Erwartung wäre die Warning müsste eigentlich immer kommen.

> der zweite Grund, warum ich keinen PR aufgemacht habe - ich gehe/ging trotzdem davon aus, daß Du den Branch gemeldet bekommen hast vom GitHub

Weiß nicht, was genau Du unter "von GitHub gemeldet bekommen hast" meinst. Ich habe weder im GitHub-WebIf noch per Mail eine Benachrichtigung bekommen, @PeterPawn hat einen Branch mit dem Namen so-und-so aufgemacht. Bzw. habe sowas noch nie bekommen.

z. Info: die Benachrichtigungen per Mail habe ich vor einiger Zeit abgeschaltet, da diese unstrukturiert kamen und irgendwann mal nur noch genervt haben.

closed 2019-07-02 by er13