Freetz/freetz #207: 7390 intl.: missing setting for TAM plugin integration

PeterPawn, 2019-07-06:
Starting (at least) with version 06.83, the international firmware for 7390 devices contains a non-empty TAM plugin, too:

```
~ $ unsquashfs -s plugin-tam.image
Reading a different endian SQUASHFS filesystem on plugin-tam.image
Found a valid big endian SQUASHFS 3:76 superblock on plugin-tam.image.
Creation or last append time Thu Mar 23 15:27:23 2017
Filesystem size 502.80 Kbytes (0.49 Mbytes)
Compression lzma
Block size 65536
Filesystem is exportable via NFS
Inodes are compressed
Data is compressed
Fragments are compressed
Always-use-fragments option is not specified
Check data is not present in the filesystem
Duplicates are removed
Number of fragments 7
Number of inodes 390
Number of uids 1
Number of gids 0
```

Version 84.06.53 intl. didn't contain this plug-in - I don't know for sure, whether there were other versions in-between.

---

My proposed change removes the restriction "only the german version" from the condition:

```
diff --git a/config/avm/features.in b/config/avm/features.in
index 24d240ce3..a543a4d6d 100644
--- a/config/avm/features.in
+++ b/config/avm/features.in
@@ -220,7 +220,7 @@ config FREETZ_AVM_HAS_PLUGIN_TAM
        default y
        depends on \
                (FREETZ_TYPE_7270_V1                        && FREETZ_AVM_VERSION_04_XX_MAX) || \
-               (FREETZ_TYPE_7390    && FREETZ_TYPE_LANG_DE && FREETZ_AVM_VERSION_06_8X_MIN)
+               (FREETZ_TYPE_7390                           && FREETZ_AVM_VERSION_06_8X_MIN)

 config FREETZ_AVM_HAS_PLUGIN_WEBCM_INTERPRETER
        depends on FREETZ_AVM_HAS_PLUGINS_UPDATE
```

er13, 2019-07-06:
Looks good/plausible. Would you like to submit a PR? Or should I intergrate your changes manually?

PeterPawn, 2019-07-06:
Too small changes for a new branch - it's only a small fix, that may be applied without any risk (as far as I can see) directly to the master branch, if it does not require any further testing.

closed 2019-07-06 by er13