Freetz/freetz #208: strace: problem compiling package for 7390 since version bump to 5.0

PeterPawn, 2019-07-12:
Seit dem Update auf Version 5.0 des `strace`-Pakets, läßt sich dieses nicht mehr für die 7390 kompilieren, da dort von AVM in den Kernel-Quellen eine Definition für die Socket-Option `SO_PROTOCOL` als 38 erfolgt(e) und die Version 5.0 für MIPS-Architekturen erwartet, daß diese Nummer 4136 wäre:

```
#if defined __alpha__ || defined __hppa__ || defined __mips__ || defined __sparc__
SO_PROTOCOL     4136
#else
SO_PROTOCOL     38
#endif
``` 

(aus `xlat/sock_options.in` in den `strace`-Quellen)

und das auch mit entsprechenden Assertions sicherstellen will.

Nun gibt es wieder mehrere Wege, um für ein Freetz-Paket einen zusätzlichen Patch in Abhängigkeit vom verwendeten Modell und/oder der verwendeten Kernel-Version einzufügen ... daher schenke ich mir hier auch den Vorschlag, wie man das - generisch - lösen könnte, so daß die Freetz-Quellen einheitlich bleiben. Es wäre ja sogar denkbar, daß man für die 7390 weiterhin die 4.9 verwendet, auch wenn die einen Kernel 2.6.28 benutzt und keine 2.6.19.

Für meinen eigenen Bedarf habe ich als Umgehungslösung einfach einen zusätzlichen Patch für die 7390 aufgenommen (davon ausgehend, daß die keine 2.6.19 verwendet und damit die Paketversion immer 5.0 ist bei der 7390 - ansonsten paßt der Patch nicht zu den `strace`-Quellen), der im Branch `strace` meines Forks nachgelesen werden kann (https://github.com/PeterPawn/YourFreetz/commit/ac7857ae1555a8fb82a53b082956bc56ed9ac2cf). 

Möglicherweise kann man das genauso an der Kernel-Version festmachen, wie die Unterscheidung in 4.9 und 5.0 - da ich bei mir per Definition max. bis zur 7390 zurück arbeiten will und nur die neuesten Kernel für jedes Modell unterstütze, reicht mir persönlich die umgesetzte Lösung ebenfalls.

Theoretisch benutzen zwar ausschließlich die Vx180-Boxen (also primär die 7390) den Kernel 2.6.28, aber bei AVM liest sich das jedenfalls an der Stelle, wo `SO_PROTOCOL` in `socket.h` (für MIPS) bei der 7390 definiert ist, so (Quellen für 06.83):

```
 79 #define SO_PROTOCOL             38  /* Backport 2.6.32 */
 80 #define SO_DOMAIN               39  /* Backport 2.6.32 */
```

Das läßt bei mir die Frage offen, ob es noch 2.6.32-Versionen von AVM gab (vor 2.6.32.61), wo ebenfalls die 38 für `SO_PROTOCOL` gesetzt wird - bei der 7490 (06.30-Quellen) ist das schon mal nicht der Fall, ebensowenig bei den 05.51-Quellen für die 7270. 

Aber irgendwoher muß diese "Backport"-Idee ja kommen, wenn das nicht am Ende nur ein "Mißverständnis" bei AVM war und jemand dort schlicht die Definition für die falsche Architektur in die 2.6.28 übernommen hat. Wobei dann aus der bei MIPS üblichen Nummer - und zwar wohl schon von Beginn an in der 2.6.32 - 4136 (0x1028) die 38 (0x26) von anderen Plattformen wurde. Solange alles mit derselben Definition arbeitet, fällt es halt nicht weiter auf.

Vielleicht ist es ja doch am besten, wenn man die 4.9 von `strace` bis einschließlich zur 2.6.28 "festtackert" ... dann braucht man sich zumindest an dieser Stelle auch keine Gedanken über einen Patch für `strace` zu machen und eigentlich ist die 7390 ja auch schon tot. Ich unterstütze die bei mir auch nur noch aus Interesse und weil sie halt ein Modell mit NOR-Flash ist, das ich auch tatsächlich besitze (jedenfalls noch zwei Boxen für solche Experimente) - im Gegensatz zu den neueren Modellen mit NOR-Flash und kombiniertem Kernel+Dateisystem-Layout.

er13, 2019-07-14:
Hmm, works for me
```
mkdir -p packages/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/strace-5.0/root
if test -d make/strace/files; then 	tools/tar-gnu -cf - -C make/strace/files --exclude=.svn --exclude=.git --exclude=.gitignore --exclude=.build-prereq-checked --exclude=.unpacked --exclude=.configured --exclude=.compiled --exclude=.installed . | tools/tar-gnu -xf - -C packages/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/strace-5.0; fi
---> package/strace: preparing... mkdir -p source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/strace-5.0; tools/unxz -c dl/strace-5.0.tar.xz | tools/tar-gnu -x -C source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/strace-5.0 --transform='s|^./\+||' --strip-components=1 
set -e; shopt -s nullglob; for i in make/strace/patches/*.patch* make/strace/patches/5.0/*.patch*; do tools/freetz_patch source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/strace-5.0 $i; done; 
    applying patch file make/strace/patches/5.0/010-no__dm_ioctl_data__in_2_6_19.patch
    patching file dm.c
    ----------------------------------------------------------------------
    applying patch file make/strace/patches/5.0/020-missing_include_in__linux_audit_h__in_2_6_19.patch
    patching file netlink.c
    patching file printsiginfo.c
    ----------------------------------------------------------------------
    applying patch file make/strace/patches/5.0/030-ipv6_stuff_redefined_workaround.patch
    patching file rtnl_mdb.c
    ----------------------------------------------------------------------
    applying patch file make/strace/patches/5.0/040-different_key_codes_in_patched_2.6.19_used_by_AVM.patch
    patching file xlat/evdev_keycode.in
    patching file xlat/evdev_keycode.h
    ----------------------------------------------------------------------
configuring... (conf_cmd() { ./configure  "$@"  || { printf "\n\\033[33m%s\\033[m\n" "ERROR: Build failed.";  exit 1; } }; cd source/target-mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/strace-5.0;  rm -f config.{cache,status};    PATH="/freetz/freetz-dev/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/bin:/freetz/freetz-dev/toolchain/build/mips_gcc-4.8.5/mips-unknown-linux-gnu/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/gene/local/bin" FREETZ_LIBRARY_DIR="/usr/lib/freetz" FREETZ_KERNEL_VERSION_MAJOR="2.6.28" FREETZ_TARGET_ARCH="mips" FREETZ_TARGET_ARCH_ENDIANNESS_DEPENDENT="mips" CC="/freetz/freetz-dev/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/bin/mips-linux-uclibc-gcc" CXX="/freetz/freetz-dev/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/bin/mips-linux-uclibc-g++-wrapper" CFLAGS="-march=24kc -mtune=24kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64" CXXFLAGS="-march=24kc -mtune=24kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64" LDFLAGS="" PKG_CONFIG_PATH="/freetz/freetz-dev/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/bin/../lib/pkgconfig" PKG_CONFIG_LIBDIR="/freetz/freetz-dev/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/bin/../lib/pkgconfig" GLOBAL_LIBDIR=/freetz/freetz-dev/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/usr/lib ac_cv_header_linux_netlink_h=yes CONFIG_SITE=/freetz/freetz-dev/include/config.site/mips-linux-uclibc conf_cmd   --cache-file=/dev/null --target=mips-linux --host=mips-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --datadir=/usr/share --includedir=/usr/include --infodir=/usr/share/info --libdir=/usr/lib --libexecdir=/usr/lib --localstatedir=/var --mandir=/usr/share/man --sbindir=/usr/sbin --sysconfdir=/etc --with-gnu-ld --disable-nls    )
configure: WARNING: unrecognized options: --with-gnu-ld, --disable-nls
configure: loading site script /freetz/freetz-dev/include/config.site/mips-linux-uclibc
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for mips-linux-strip... mips-linux-strip
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking build system type... x86_64-pc-linux-gnu
checking host system type... mips-unknown-linux-gnu
checking for mips-linux-gcc... /freetz/freetz-dev/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/bin/mips-linux-uclibc-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether /freetz/freetz-dev/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/bin/mips-linux-uclibc-gcc accepts -g... yes
checking for /freetz/freetz-dev/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/bin/mips-linux-uclibc-gcc option to accept ISO C89... none needed
checking whether /freetz/freetz-dev/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/bin/mips-linux-uclibc-gcc understands -c and -o together... yes
checking for style of include used by make... GNU
[... cut ...] || echo './'`rtnl_neigh.c
mv -f .deps/strace-rtnl_neigh.Tpo .deps/strace-rtnl_neigh.Po
/freetz/freetz-dev/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/bin/mips-linux-uclibc-gcc -std=gnu99 -DHAVE_CONFIG_H   -I./linux/mips -I./linux/mips -I./linux -I./linux -I. -I.      -Wall -Wempty-body -Wformat-security -Wignored-qualifiers -Winit-self -Wlogical-op -Wmissing-parameter-type -Wnested-externs -Wold-style-declaration -Wold-style-definition -Woverride-init -Wsign-compare -Wt

er13, 2019-07-14:
Bitte die `.config` zur Verfügung stellen und den eigentlichen Fehler präsentieren - dieser ist nämlich nirgends zu sehen, nur Deine Lösung dafür.

er13, 2019-07-14:
U. U. ist nur die self-built toolchain betroffen, mein Test erfolgte mit der precompiled-toolchain. Wenn dies der Fall ist, reicht es den fehlerhaften Backport aus den AVM-Kernel-Quellen für 06.83 wieder rauszupatchen.

PeterPawn, 2019-07-14:
[config.compressed.txt](https://github.com/Freetz/freetz/files/3389719/config.compressed.txt)
[make-strace.txt](https://github.com/Freetz/freetz/files/3389723/make-strace.txt)

> U. U. ist nur die self-built toolchain betroffen, mein Test erfolgte mit der precompiled-toolchain. Wenn dies der Fall ist, reicht es den fehlerhaften Backport aus den AVM-Kernel-Quellen für 06.83 wieder rauszupatchen.

Wenn AVM den Kernel und alles andere für die 7390 selbst aber mit 38 für `SO_PROTOCOL` (`SO_DOMAIN` ist ebenfalls betroffen) anstelle von 4136 übersetzt hat, unterdrückt dieser Vorschlag (wenn ich ihn richtig verstehe) zwar das Problem beim Kompilieren, führt aber bei der Ausführung des solchermaßen übersetzten `strace` zu Fehlinterpretationen bei Socket-Aufrufen auf der 7390.

er13, 2019-07-14:
Die `SO_PROTOCOL` / `SO_DOMAIN` related Backports sind in 7390.06.5x hinzugekommen.

[Für ARM](https://github.com/Freetz/freetz/blob/1b01b3b2aa080c149777aac17361f2d203dffd15/make/linux/patches/2.6.28.10/7390_06.51/002-7390.06.30-7390.06.51-delta.patch#L12) - richtig.

[Für MIPS](https://github.com/Freetz/freetz/blob/1b01b3b2aa080c149777aac17361f2d203dffd15/make/linux/patches/2.6.28.10/7390_06.51/002-7390.06.30-7390.06.51-delta.patch#L2066) - falsch. Diese hätten für MIPS wie folgt lauten müssen:
```
#define SO_PROTOCOL 0x1028  /* protocol type */
#define SO_DOMAIN   0x1029  /* domain/socket family */
```

Leider werden die `#define's` referenziert - [hier](https://github.com/Freetz/freetz/blob/1b01b3b2aa080c149777aac17361f2d203dffd15/make/linux/patches/2.6.28.10/7390_06.51/002-7390.06.30-7390.06.51-delta.patch#L7756) und [hier](https://github.com/Freetz/freetz/blob/1b01b3b2aa080c149777aac17361f2d203dffd15/make/linux/patches/2.6.28.10/7390_06.51/002-7390.06.30-7390.06.51-delta.patch#L7765) - sodass rauspatchen bzw. korrigieren der IDs keine Lösung ist, wie Du richtig angemerkt hast, weil der AVM-Kernel eben mit falschen IDs übersetzt wurde.

PeterPawn, 2019-07-14:
Trotzdem ist es dann immer noch unverständlich, wieso die "precompiled toolchain" für die 7390 (die ja auch die Kernel-Header enthält oder hat sich das geändert?) dann die korrekten MIPS-Definitionen enthalten kann/soll, wenn die AVM-Quellen doch die falsche Definition verwenden. 

Das würde ja darauf hinauslaufen, daß die 7390-**Download**-Toolchain (die ist nun mal einigermaßen "unique", wenn mein Gedächtnis nicht trügt und die Fusiv-Boxen die einzigen sind, die Kernel-Version 2.6.28 verwendet haben) falsche Header-Files enthält.

Auffällig wird dieses wohl erst dadurch, daß die Version 5.0 von `strace` (die 4.9 ist 5 Jahre alt) jetzt diese Assertions nachgerüstet hat, wenn die Optionen nicht die erwarteten Nummern haben.

er13, 2019-07-14:
Die download-toolchain für 7390.06.2x - 06.5x - 06.8x ist identisch und wurde noch mit [06.2x](https://github.com/Freetz/freetz/blob/1b01b3b2aa080c149777aac17361f2d203dffd15/tools/build_download_toolchains#L48) gebaut. Daher ist die download-toolchain nicht betroffen.

Die Kernel-Headers werden nach dem Auspacken der download-toolchain in dieser NICHT aktualisiert, weil dies im Rahmen des Build-Prozesses von uClibc geschieht und die Headers dabei sowohl woanders im Vergleich zu deren Location unter `linux-$(VER)` kopiert werden (z.B. ARCH verschwindet aus der Verzeichnis-Hierarchie) als auch die Headers an sich preprocessed werden.

PeterPawn, 2019-07-14:
Was dann im Endergebnis aber doch heißt, daß für die 7390 kompilierte Programme seit der 06.5x (oder seit wann sonst? - das wären jetzt jedenfalls 3 1/4 Jahre, wenn meine "Annahme" zeitlich hinkommt) für `SO_PROTOCOL` und `SO_DOMAIN` die falschen Nummern verwendet haben?

Wenn die Toolchain noch für 06.2x erstellt wurde und seit 06.5x die (falschen) Definitionen der Socket-Options als Backport von der 2.6.32 bei AVM in den Kernel-Quellen sind, dann würden ja seit der 06.5x entsprechende Programme mit der "falschen" Nummer (4136) arbeiten, zumindest wenn sie mit dem AVM-Kernel und anderen AVM-Programmen zusammenarbeiten sollen.

Nun sind das wohl nicht soo viele, die SO_PROTOCOL überhaupt referenzieren, weil sie mittels "getsockopt()" anhand eines vorhandenen Handle den Protocol-Type bzw. die Domain vom verwendeten "socket()"-Call (https://linux.die.net/man/2/socket) **nachträglich** ermitteln müssen, weil sie den "socket()"-Call nicht selbst ausgeführt haben oder Handle mit verschiedenen Optionen von demselben Code-Teil verarbeitet, aber unterschiedlich behandelt werden sollen und es keine weiteren Flags parallel gibt, die z.B. die IP-Family (AF_INET bzw. AF_INET6) auseinanderhalten. Die beiden Optionen sind ja ohnehin "read-only" und spielen damit nur bei einem "getsockopt()"-Call überhaupt eine Rolle.

Bei meiner Paketauswahl oben (https://github.com/Freetz/freetz/files/3389719/config.compressed.txt) betrifft das jedenfalls nur das `strace`-Paket und die `glib2`.

er13, 2019-07-14:
Ja, Deine Annahme in Bezug auf `strace` wäre vermutlich richtig, klingt jedenfalls für mich plausibel. Bei den anderen Paketen müsste man erst nachschauen, wie genau sie sich verhalten. In 7390.06.2x hat es die `SO_PROTOCOL/SO_DOMAIN`-Symbole gar nicht gegeben, die download-Toolchain ethält die `#defines's` also gar nicht. Wenn die Pakete sowas wie `ifdef(SO_PROTOCOL/SO_DOMAIN)` enthalten, dann werden die beiden SOCKET_OPTIONS einfach gar nicht behandelt, was etwas anderes ist als "behandelt mit falschen Codes".

Ansonsten finde ich nichts daran verkehrt, dass dieselbe download-toolchain für mehrere Firmware-Versionen verwendet wird. 7390.06.2x/06.5x/06.8x basieren alle auf 2.6.28.10 - da ist grundsätzlich nicht zu erwarten, dass irgendwelche `syscalls` hinzukommen bzw. dass sich irgendwelche `#define's` auf inkompatible Art ändern. Auch ein richtiger Backport wäre kein Problem. Erst ein falscher Backport verursacht ein Problem und sowas erwartet man nun mal nicht.

PeterPawn, 2019-07-14:
Gut ,,, ich baue ohnehin immer selbst die Toolchains (schon weil ich eine gänzlich andere Linux-Distro verwende und da mit etwas Pech schon die C-Library nicht zur Toolchain paßt, die mit Ubuntu erstellt wurde) und daher kann/will ich keine "Haltung" zum Thema "Download-Toolchain" haben. Meine Meinung, wo da innerhalb von Freetz die Grenzen verlaufen (sollten), ist ja bekannt.

Spannend ist nur, wie man das jetzt löst bzw. ob man das überhaupt angehen will. Die 7390 ist m.E. eben schon "tot" ... also braucht's vermutlich auch keine aktuelle `strace`-Version dafür. Ansonsten gibt es sicherlich auch noch Freetz-Benutzer, die bei der verwendeten Konstellation (7390, 06.85, eigene Toolchain + den Wunsch, das `strace`-Paket zu bauen, um z.B. die AVM-Firmware genauer zu analysieren - ja, ich weiß, daß mit jedem weiteren Kriterium die Menge der potentiell Betroffenen kleiner wird) sich nicht selbst zu helfen wissen ... wenn man also "alles" unterstützen will (alte Box, eigene Toolchain, neues `strace`), muß man wohl *irgendetwas* unternehmen - notfalls eben (wie oben schon angetextet) auch für die 7390 dann eine alte `strace`-Version verwenden.

Für mich ist das auch nur deshalb von Bedeutung, weil ich den Branch (strace) im Fork gerne loswerden würde und den Freetz-Fork trotzdem als Build-Umgebung für von mir bereitgestellte Binaries (die gibt's in Kürze auch noch für die 7390, wenn ich alles beisammen habe - selbst wenn da `strace` nicht zwingend dabei ist) nutzen würde, um die GPL-Lizenzbedingungen zu erfüllen. 

Wenn das am Ende auf Version 4.9 hinausläuft, ist mir das auch einigermaßen egal - Hauptsache ist, daß es funktioniert. Die (korrekte) Erkennung, ob ein (IP-)Socket nun über eine IPv4- oder IPv6-Adresse aufgebaut wurde, kann in so einer Ausgabe von `strace` allerdings schon interessant sein, wenn man einem Fehler hinterher rennt - und das würde dann wohl die 5.0 erfordern (wobei `strace` inzwischen ohnehin schon bei 5.2 angekommen ist: https://strace.io/files/5.2/).

er13, 2019-07-14:
Zu Dokumentationszwecken (damit man nach dem Fehler googeln kann):
```
/freetz/freetz-dev/toolchain/build/mips_gcc-4.8.5_uClibc-0.9.33.2-nptl/mips-linux-uclibc/bin/mips-linux-uclibc-gcc -std=gnu99 -DHAVE_CONFIG_H   -I./linux/mips -I./linux/mips -I./linux -I./linux -I. -I.     -Wall -Wempty-body -Wformat-security -Wignored-qualifiers -Winit-self -Wlogical-op -Wmissing-parameter-type -Wnested-externs -Wold-style-declaration -Wold-style-definition -Woverride-init -Wsign-compare -Wtype-limits -Wwrite-strings   -march=24kc -mtune=24kc -msoft-float -Os -pipe -Wa,--trap -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -MT strace-net.o -MD -MP -MF .deps/strace-net.Tpo -c -o strace-net.o `test -f 'net.c' || echo './'`net.c
In file included from xlat/socktypes.h:4:0,
                 from net.c:57:
./static_assert.h:20:25: error: static assertion failed: "SO_PROTOCOL != 4136"
 #  define static_assert _Static_assert
                         ^
xlat/sock_options.h:891:1: note: in expansion of macro 'static_assert'
 static_assert((SO_PROTOCOL) == (4136), "SO_PROTOCOL != 4136");
 ^
./static_assert.h:20:25: error: static assertion failed: "SO_DOMAIN != 4137"
 #  define static_assert _Static_assert
                         ^
xlat/sock_options.h:908:1: note: in expansion of macro 'static_assert'
 static_assert((SO_DOMAIN) == (4137), "SO_DOMAIN != 4137");
 ^
make[3]: *** [strace-net.o] Error 1
```

closed 2019-07-14 by er13