Freetz/freetz #210: tor-0.3.5.8.tar.gz kann nicht heruntergeladen werden

msebald, 2019-07-31:
Mit der aktuellsten Version, ausgewähltem Tor Paket sowie 7590 und Firmware 7.0x kann das Torpaket nicht geladen werden:

```
Download failed - "https://www.torproject.org/dist/tor-0.3.5.8.tar.gz"  ->  error code 4
Download failed - "http://freetz.magenbrot.net/tor-0.3.5.8.tar.gz"  ->  error code 8
Download failed - "http://freetz.3dfxatwork.de/tor-0.3.5.8.tar.gz"  ->  error code 8
Download failed - "http://freetz.wirsind.info/tor-0.3.5.8.tar.gz"  ->  error code 8
```

PeterPawn, 2019-07-31:
```
freetz@zbox:~> wget -q -O - https://www.torproject.org/dist/tor-0.3.5.8.tar.gz | tar -t -z
tor-0.3.5.8/
tor-0.3.5.8/scripts/
tor-0.3.5.8/scripts/maint/
tor-0.3.5.8/scripts/maint/checkOptionDocs.pl.in
tor-0.3.5.8/scripts/maint/checkIncludes.py
[...]
```

msebald, 2019-08-01:
Warum auch immer, nun geht es. Error 4 ist DNS? Dann hatte ich nen Namensauflösungsproblem mit torproject.org und die anderen drei Domains haben das Paket nicht.

msebald, 2019-08-01:
Danke!

closed 2019-08-01 by msebald

PeterPawn, 2019-08-01:
https://www.gnu.org/software/wget/manual/html_node/Exit-Status.html#Exit-Status

>Danke!

Da nich' für.