Freetz/freetz #216: (7170) Rewrite call engine to dial single digit numbers

programminghoch10, 2019-08-09:
Hello,
I got a special situation which i faced lately. 
I want to set up an in-house phone system using an asterisk server (which already runs without problems) and several 7170 for analog to sip interfacing. The System already works flawlessly but because of old analog telephones calling numbers takes very long. I read somewhere that phone numbers with at least 3 digits get dialed through (for the box "out into the world", using SIP). My tests confirmed that. My System has about 10 internal telephones so i would use 001 - 010 but dialing zeros takes about 5 seconds each, which sums up to max 15 seconds just calling an internal number. The system previously in use would just call instantly after one digit, which held calling even zeros at low waiting times.

So long story short, is it possible to mod the internal calling program of the box to dial SIP after one digit?

Those boxes shall only furfill this one task, so removing Wifi if needed won't hurt.

Greetings, programminghoch10

er13, 2019-08-09:
Sorry, zum einen rein sprachlich nix verstanden. Und zum anderen handelt es sich um ein Bug-Tracking-System und kein Support-/Diskussionsforum. Versuch' Dein Problem etwas verständlicher auszuformulieren und versuch' Dein Glück im [IPPF](https://www.ip-phone-forum.de/forums/avm.361/).

closed 2019-08-09 by er13

programminghoch10, 2019-08-11:
Ich verstehe...
Momentan bevorzuge ich im generellen GitHub, da ich dem IPPF erst noch betreten muss.
Das mit dem sprachlich nicht verstanden ist aber ein Problem dass ich lösen werde, ich werde einfach alles neu schreiben auf Deutsch.

er13, 2019-08-11:
Als Anhaltspunkte für die (eigene) Suche:

https://www.ip-phone-forum.de/threads/problem-zwei-fb-7270-via-voip-verbunden-hinter-einer-tk-ananlage.254524/#post-1874086

https://www.ip-phone-forum.de/threads/2-stellige-externe-rufnummern-%C3%BCber-fritzbox-und-isdn-telefon.256576/

programminghoch10, 2019-08-12:
Das klingt jetzt leicht so als hätte ich nicht selber schon gesucht... Ich weiß es ist nervig wenn Leute nicht davor suchen.., Ich hab das aber durchaus gemacht. Was die 2 Links von @er13 betreffen, die sind aus 2012/13 und behandeln die Firmware 5.x .
Dennoch bleibt meine Frage, ist es möglich die Box mittels modding dazu zu verleiten auch über VOIP einzelne Ziffern zu wählen?
In sämtlichen Foren wird nur besprochen wie man das Problem umgeht, aber nirgends wird/wurde probiert ob man das mit modding lösen kann.
Außerdem möchte ich (nochmals) anmerken, dass einfach 2 ziffern davor oder hinterhängen keine Option ist, da die Wählzeiten dadurch immens verlängert werden. Auch unterstützen diese Telefone kein * oder # , weshalb ich auch nicht das beenden des Wählens mit einem # erzwingen kann.
Ich hab letztens etwas von Wahlregeln gelesen. Ich kann es zwar erst morgen testen, aber ich glaube dass auch diese mich nicht zum Erfolg bringen werden.
Wenn es an Zeit mangelt, kann mir vielleicht einer der Devs einen Anhaltspunkt geben wo ich im Code suchen müsste um selber zu versuchen zu modden?

er13, 2019-08-12:
> Dennoch bleibt meine Frage, ist es möglich die Box mittels modding dazu zu verleiten auch über VOIP einzelne Ziffern zu wählen?
Rein interessehalber: was verstehst du an der Stelle unter Modding? Patchen der AVM-Binaries/-Libraries? Es wird aber vermutlich in deinem Fall notwendig sein.

programminghoch10, 2019-08-12:
Wenn nötig dann das wohl auch... 
Wie schwer wird das sein? Ist das überhaupt möglich binaries zu patchen?

er13, 2019-08-12:
Binaries zu patchen, sofern kein neuer Code hinzugefügt werden muss, wäre möglich. Die zu patchende Stelle zu finden, ist mit einem enormen Aufwand verbunden, das zuständige Binary bzw. die zuständige Library zu identifizieren, den Code zu disassemblieren, die Anpassung zu erarbeiten...

Es ist vom zeitlichen Aufwand / Kosten her, günstiger einfach die neue Hardware zu kaufen.

programminghoch10, 2019-08-21:
Nur noch eine kurze Frage, wäre es möglich die Einträge des internen Telefonbuchs zu modifizieren und somit die Sterne weg zu bekommen und einstellige Nummern festzulegen?