Freetz/freetz #219: (patch) - php5.6.40 , fix dynamic libs und Opcache

arjunae, 2019-08-14:
Hey beisammen,

Wollte vergangene Woche php mit opcache auf einer Box einsetzen.
Nach einiger Sucherei im Web kam ans Licht, das das php build system probleme in Cross-Kompilierungs Umgebungen hat. Deswegen funktionierte weder das nachladen von php libs, noch der php opcache.

Habe dann einen alten busybox pst entdeckt, in dem die Leute das Problem fixen konnten:
http://lists.busybox.net/pipermail/buildroot/2015-April/124564.html

Habe den Patch dann getestet , keine Probleme mehr festgestellt und einen freetz patch daraus erstellt. (siehe Anhang). Ist mein erster , also kanns sein das die ein oder andere kleinigkeit noch verändert werden soll.

rockOn,
Thorsten
[freetz_dynlibs_opcache_php-5.6.40.zip](https://github.com/Freetz/freetz/files/3500562/freetz_dynlibs_opcache_php-5.6.40.zip)

er13, 2019-08-17:
Könntest Du bitte Deine Änderungen nicht als "Patch im Anhang", sondern als Pull-Request zukommen lassen? Zum einen wird dadurch dokumentiert, dass die Änderungen von Dir stammen. Und zum anderen macht es den Review-Prozess einfacher.

Könntest Du bitte die Änderungen auch thematisch auf mehrere Pull-Requests aufteilen, sowas wie
- version bump auf 5.6.40 (diese könnte schnell gemerged werden, während über den Rest noch u.U. diskutiert wird)
- fix dynamic libs (bitte bei diesem das Problem beschreiben, ich habe es bisher nicht verstanden)
- Opcache support
- etc.

Danke!

arjunae, 2019-08-17:
Hallo er13,
Ja, der version bump sollte kein größeres Problem sein :)
Die Motivation hinter dem "dynamix library fix" ist an sich auch einfach erklärt:
php gibt beim versuch Zend extensions wie z.b suhoshin oder auch opchache folgenden fehler aus:
`php extensions are not supported on this platform`
 Der Grund dafür ist, das das php configure script in cross-compile umgebungen nicht richtig arbeitet und die variablen ac_cv_*_dlopen nicht setzt. Das führte leider später dazu, das in zend.h die extensions deaktiviert  und so nicht nutzbar wurden.

Der zweite Punkt betrifft die Kompilierung des opcache moduls im cross-compile Modus.
Ohne den patch gibts beim laden blos einen segfault.
Die leute vom busybox projekt haben deshalb Änderungen an den autoconfig files des opcaches implementiert und ein paar CFlags hinzugefügt.

Was meinst Du  - Ein Patch für den bump und ein zweiter, damit Zend Extensions und Opcache funktionieren ? 

cheers, Thorsten

er13, 2019-08-17:
Keine Patches bitte, Pull-Requests. Und ansonsten bitte so, dass jede Änderung in sich abgeschlossen und vollständig ist, sodass man sie leichter bzw. überhaupt reviewen kann:
- version bump
- fix dynamic libs aka Zend extensions support
- add Opcache support
- da ist meiner Ansicht nach noch etwas, was ich nicht verstehe wofür es da ist

arjunae, 2019-08-18:
Oki, machen wir. :)  
`PKG_PREVENT_RPATH_HARDCODING` funktionierte zwar mit 5.6.37, führte bei 5.6.40 jedoch zu nicht auflösbaren runtime verweisen. Wenns dafür ne andere Möglichkeit gibt, klar, gerne. 
Werde dann mal den aktuellen trunk forken.  
vg!

closed 2019-12-11 by arjunae