Freetz/freetz #226: Support for German Freifunk

JasMich, 2019-09-07:
Are you generally interested in integrating the German Freifunk packages into freetz? This would allow using old AVM router to connect via `B.A.T.M.A.N.` or `gluon` to the Freifunk mesh.

heini66, 2019-09-10:
it's a little bit more than batman... and to supprt gluon. ( gluon is a openwrt based framework and won't work with freetz )
first of all, you have to get 8023.11s working on the wifi. if i'm right, the wifi driver *.ko ist avm stuff and won't do 11s. so you have to replace the wifi ko to get 11s working and you will loose the option to edit wlan settings via the avm gui.
next point ist go get the respondd workinf to get data showing up the node an the community map.

my 2 cent's are to have a look @the openwrt tragets and create gluon suport for them.
https://github.com/freifunk-gluon/gluon/pull/1803 - 05

er13, 2019-10-30:
Feel free to implement it yourself and provide a pull-request, please however don't expect any support from us (no time, no know-how).

closed 2019-10-30 by er13