Freetz/freetz #227: Reboot when connecting OpenVPN

Helmut1972, 2019-09-13:
Hi,

freetz-master-20190819-baa9a5e54 on 7490

Using OpenVPN when connecting the box reboots:

Sep 13 14:53:44 ipv4-gw. kernel: [ 2705.970000][0][0]system-load 100% loadavg 0.31 0.19 0.14 - 122 tasks:10 % curr:openvpn(0 %) max:migration/0(8 %, pid:7) pgstat: sum=62373 free=21827 slab=4875 alloc=65/s fault=173/s (sleep 1)
Sep 13 14:53:55 ipv4-gw. kernel: [ 2717.580000][0][kernel-trap] pc=0x804da8c8(0x804da8c8 __netif_receive_skb+0x8/0xa0) addr=0x00000001 task=openvpn pid=5781 ra=0x804dbc5c(0x804dbc5c process_backlog+0xbc/0x220)
Sep 13 14:53:55 ipv4-gw. kernel: [ 2717.580000][0]Code(0x804da8c0): 0x00801021 0x8c830010 <0x00030336> 0x3c038099 0x8c6312c0
Sep 13 14:53:55 ipv4-gw. kernel: [ 2717.580000][0]set_reboot_status: Soft-Reboot(KCRASH)  -  KCRASH(4) SUM(4)
Sep 13 14:53:55 ipv4-gw. kernel: [ 2717.580000][0]BUG_ON((unsigned long)(skb->sk)) at function '__netif_receive_skb' line: 3706 file: /GU/RELEASE_KERNEL_vr9_build/linux/net/core/dev.c
Sep 13 14:53:55 ipv4-gw. kernel: [ 2717.580000][0]Kernel bug detected[#1]:
Sep 13 14:53:55 ipv4-gw. kernel: [ 2717.590000][0]CPU: 0 PID: 5781 Comm: openvpn Tainted: P           O 3.10.107 #1
Sep 13 14:53:55 ipv4-gw. kernel: [ 2717.590000][0]task: 8fe2bb60 ti: 8d304000 task.ti: 8d304000

Helmut1972, 2019-09-13:
[config.txt](https://github.com/Freetz/freetz/files/3610174/config.txt)

PeterPawn, 2019-09-13:
Load "yf_patchkernel.ko", before you try to open a TUN/TAP device.

https://github.com/Freetz/freetz/pull/66

Helmut1972, 2019-09-13:
Danke, wo, im Interface unter Freetz -> Moules oder beim Bauen?

PeterPawn, 2019-09-13:
Beispiel für das "openvpn-v2-cgi" gibt's im oben verlinkten PR, bei der Aktivierung von OpenVPN in der Freetz-Konfiguration sollte das LKM automatisch mit ins Freetz-Image kopiert werden bei den passenden Modellen ... heißt: Beim Zusammenstellen des Images kümmert sich Freetz schon drum, nur das (ältere) OpenVPN-GUI lädt das LKM nicht automatisch (v2 dann eben schon).

Helmut1972, 2019-09-14:
Hat alles geklappt, Box ist stabil. Danke!

closed 2019-10-30 by er13