Freetz/freetz #230: Unhide Annex has no effect on FB 7312

franknord, 2019-10-16:
Hello,

Actual:
* Using master (71f6483922ffc067d20edb96e87f83f9ee696d01 - as of today, Fritz OS 6.02 / 6.03 and 6.05 / 6.06) on a FB 7312 has no effect. No option is shown - neither at "Störsicherheit" nor in the setup wizzard.

Expected:
* I'm able to switch the annex (A/B) at both locations.

PeterPawn, 2019-10-16:
Does it work with the current (latest) version 06.55?

What's the output of

```
grep MULTI_ANNEX /etc/init.d/rc.conf
```

from a shell prompt at the device?

If the patch was applied correctly, all statements with `export CONFIG_DSL_MULTI_ANNEX="n"` should only be comments anymore.

In this case, further investigations are necessary, whether the code has changed from 06.0x to 06.5x - I've provided (and tested) the patch in `modfs` only for later versions and the Freetz patch was inspired by this one.

The simplest (and only suspected) reason could be a different `export`-statement in `rc.conf` (e.g. missing or single quotes around the `n` value) - that's why the result of the patch has to be checked first.

PeterPawn, 2019-10-16:
Looks odd ... the file `/etc/init.d/rc.conf` from 06.55 for 7312 (afaik there's only one single "flavor" of this firmware and no international (nor any other) version was published) contains (originally) the following occurrences of `MULTI_ANNEX` ... and the patch (from Freetz) doesn't _remove_ any of these lines:

```
vidar:/tmp/7312_06.55 $ grep MULTI_ANNEX etc/init.d/rc.conf
export CONFIG_DSL_MULTI_ANNEX="y"
export CONFIG_DSL_MULTI_ANNEX="n"
export CONFIG_DSL_MULTI_ANNEX="n"
if [ "${CONFIG_DSL_MULTI_ANNEX}" = "y" ] ; then
vidar:/tmp/7312_06.55 $
```

Please attach any files (screenshots and even the configuration file, you've used) directly to this issue - so they're kept, regardless of any later removal from the Heinlein servers.

Therefore I didn't look into the provided screenshots, but I do not expect any new info out of them, too. If an item is missing, the screenshot simply shows this - but no more info, *why* it's not displayed.

You may look (using a shell prompt) for the final value of `CONFIG_DSL_MULTI_ANNEX` (in the shell environment), after `rc.conf` was processed - the HTML elements for a selection of the spec-annex for DSL bandplans are only generated, if this value is `y`. Any other value doesn't reveal this settings - that's the code from `/usr/www/avm/internet/dsl_line_settings.lua`:

```lua
228 function write_multi_annex_visible()
229 if config.DSL_MULTI_ANNEX then
230 box.out([[<hr><h4>]])
231 box.html([[{?5085:755?}]])
232 box.out([[</h4><div class="formular">]])
233 box.out([[<input id="uiAnnexA" type="radio" name="annex" value="A" ]], get_annex_checked("A"), [[><label for="uiAnnexA">]])
234 box.html([[{?5085:195?}]])
235 box.out([[</label><br>]])
236 box.out([[<input id="uiAnnexB" type="radio" name="annex" value="B" ]], get_annex_checked("B"), [[><label for="uiAnnexB">]])
237 box.html([[{?5085:419?}]])
238 box.out([[</label></div>]])
239 box.out([[<p>]])
240 box.html([[{?5085:554?}]])
241 box.out([[</p>]])
242 box.out([[<p><strong>]])
243 box.html([[{?5085:422?}]])
244 box.out([[</strong>]])
245 box.html([[ {?5085:975?}]])
246 box.out([[</p>]])
247 end
248 end
```

I'm a bit puzzled, why your `rc.conf` file looks so strange - perhaps a problem during the Freetz build?

PeterPawn, 2019-10-16:
The 7312 firmware contains only a single `CONFIG_DSL_MULTI_ANNEX="n"` and no further code to create the needed HTML elements for a GUI-based switching of the used version in `dsl_line_settings.lua`.

The Freetz patch should exclude this model (and probably all other models, where the "main" value of `CONFIG_DSL_MULTI_ANNEX`  (or the first occurrence of this setting in `rc.conf`) doesn't read `y` - then there're much more changes needed to provide this option (at least the `dsl_line_settings.lua` file needs an addition).

But the other manners to switch the `annex` value (from urlader environment or from `kernel_args`) should work as expected.

franknord, 2019-10-17:
Thanks for your help - I attached screenshots, config.txt -  please ping, in case you need the resulting firmware binary.

Device-configuration using `kernel_args` works. I created a small patch explaining the situation. It'd be great, if freetz includes an annex selection dialog - however `kernel_args` is fine for my use-case.

* [config.txt](https://github.com/Freetz/freetz/files/3738460/config.txt)
* [image]
* [image]

closed 2019-12-01 by er13

sfritz2, 2020-10-10:
@franknord  The easiest way to get Annex Selection is to flash with 7330 SL Firmware or build an 7330 SL alien for 7312 on freetz-ng.