Freetz/freetz #244: tincd causes reboot loop with Fritz OS 7.12

mnagl, 2019-11-30:
After upgrading to a current Freetz with Fritz OS 7.12, a 7490 box reboots several seconds after booting. 

# cat /proc/avm/log_sd/panic
`...
[  199.570000][1]807776a0 __func__.4154+0x112f70/0x11f2d8
[  199.580000][1]80870000 rawv6_prot+0x5c/0xcc
[  199.580000][1]807b0000 irq_desc+0x3920/0x7120
[  199.580000][1]8191abe8 [page: type:reserved]
[  199.590000][1]804dbad0 net_rx_action+0x190/0x260
[  199.590000][1]8dc2c6c0 [slab: type:skbuff_head_cache size:448 start:0x8dc2c6c0+0x0 allocated by:0x804caba0 __alloc_skb+0x60/0x1a0]
[  199.610000][1]8df57ca8 [stack(tincd): 8df54000+0x3ca8]
[  199.610000][1]8fd072c0 [slab: type:kmem_cache size:128 start:0x8fd072c0+0x0]
[  199.620000][1]807b9f4c __per_cpu_offset+0x0/0x8
[  199.620000][1]807ac640 softirq_vec+0x0/0x40
[  199.630000][1]808c0404 actual_skb_cache+0x0/0x4
[  199.630000][1]80983c10 simple_profiling+0x0/0x20
[  199.640000][1]807ac64c softirq_vec+0xc/0x40
[  199.640000][1]80053004 __do_softirq+0x184/0x300
[  199.640000][1]8df57d00 [stack(tincd): 8df54000+0x3d00]
[  199.650000][1]8df57e88 [stack(tincd): 8df54000+0x3e88]
[  199.650000][1]8df72420 [slab: type:kmalloc-8192 size:8192 start:0x8df72000+0x420 allocated by:0x804dbe2c alloc_netdev_mqs+0x6c/0x380]
[  199.670000][1]808c0588 kstat+0x0/0x30
[  199.670000][1]80980000 avm_memory_statistic+0x1da0/0x4058
[  199.680000][1]808ecae0 irq_stat+0x0/0x40
[  199.680000][1]8df57e88 [stack(tincd): 8df54000+0x3e88]
[  199.680000][1]8df72420 [slab: type:kmalloc-8192 size:8192 start:0x8df72000+0x420 allocated by:0x804dbe2c alloc_netdev_mqs+0x6c/0x380]
[  199.700000][1]8df57d48 [stack(tincd): 8df54000+0x3d48]
[  199.700000][1]Call Trace:
[  199.700000][1]3c20: [<804da8c8>] 0x804da8c8 __netif_receive_skb+0x8/0xa0
[  199.710000][1]3c20: [<804dbc5c>] 0x804dbc5c process_backlog+0xbc/0x220
[  199.720000][1]3c58: [<804dbad0>] 0x804dbad0 net_rx_action+0x190/0x260
[  199.720000][1]3c98: [<80053004>] 0x80053004 __do_softirq+0x184/0x300
[  199.730000][1]3cf8: [<80053288>] 0x80053288 do_softirq.part.1+0x68/0xa0
[  199.730000][1][1]system-load 1 loadavg 0.28 0.39 0.17 - 115 tasks:2 % curr:tincd(0 %) max:ctlmgr(0 %, pid:2884) pgstat: sum=62844 free=25850 slab=4483 alloc=47/s fault=98/s (sleep 0)
[  199.750000][1]3d10: [<804da774>] 0x804da774 netif_rx_ni+0x34/0x60
[  199.760000][1]3d30: [<8045f17c>] 0x8045f17c tun_get_user+0x8dc/0xca0
[  199.770000][1]3de8: [<8045fb10>] 0x8045fb10 tun_chr_aio_write+0x70/0xa0
[  199.770000][1]3e10: [<80122f08>] 0x80122f08 do_sync_write+0x88/0x100
[  199.780000][1]3ea8: [<80123bc0>] 0x80123bc0 vfs_write+0xc0/0x2a0
[  199.780000][1]3ed8: [<80124518>] 0x80124518 SyS_write+0x58/0x100
[  199.790000][1]3f10: [<800048b0>] 0x800048b0 stack_done+0x20/0x40
...`

PeterPawn, 2019-11-30:
With the first glance at the call trace, I had a Déjà-vu - https://github.com/Freetz/freetz/pull/66

I'd presume, there's a `modprobe` or `insmod` missing in the `tinc` package and probably the module wasn't built (due to missing dependencies), too.

mnagl, 2019-11-30:
The tinc documentation only lists tun as dependency. A tun module is indeed not available on the box, but it seems that tun is included in the kernel. At least tunctl does not complain. The reboot loop occurs even with tincd 1.1pre

PeterPawn, 2019-11-30:
The real reason for the reboot was missing from your first description - the only possible conclusion was, that there was a kernel panic due to the mentioned source of the shown error (log_sd/panic). 

If the problem is the same kernel trap, as in the (IPPF) thread, which was referred to by my pull request, the "live patches" from `yf_patchkernel.ko` should/could solve your problem.

Because the `tinc` package from Freetz doesn't rely on `tun` (and is obviously only seldom used), nobody did recognize until now, that this package is also involved by these traps.

Try to include the mentioned module (change your `.config` file and add/set `FREETZ_MODULE_yf_patchkernel=y`) into your image and add a command (to load the LKM) to the start script of the `tinc` package (prior to the line that starts the daemon).

If it runs now, the problem was (obviously) solved and you may provide (or at least suggest with more precision) a change to this (rather rarely used) package.

mnagl, 2019-12-01:
I am starting tincd from rc.custom. It seems a change to 
```
ifconfig wlan up
modprobe yf_patchkernel
tincd --config=/var/tmp/flash/tinc/net
```
really did the trick. The box is not rebooting now for several minutes.

Thank you very much for the quick help!!