Freetz/freetz #257: firmware update failed

MasterCATZ, 2019-12-27:
I just built with openvpn support 
7490_07.12-freetz-master-20191217-8d93a8fe4.en_20191228-074425.image

I have 7.12 installed but it does not recognise this as supported  firmware 
should I flash another manufacture version on and try again ? 
from memory I had to downgrade to an old dev version before to bypass this check

PeterPawn, 2019-12-28:
You may/could/should install the new firmware using the FTP server from EVA bootloader.

https://www.ip-phone-forum.de/threads/wie-verwende-ich-denn-nun-die-skript-dateien-aus-yourfritz-eva_tools.298591/

The files are (usually) present at your own build system already ... look at `tools/yourfritz/eva_tools` folder.

PeterPawn, 2019-12-28:
Looks like you're from Australia - at least the obvious (but late realized) difference of ToD values (at local time 2019-12-28 07:44 AM the image file was built and the issue here has been opened at 2019-12-27 10:02 PM GMT) seems to be an indication for (or your clock really needs an adjustment).

You should be able to translate the description from my link above to English (I'd suggest to use DeepL for such a task) and if there're any further questions after reading, don't hesitate to ask - here or (even better) in the linked bulletin board.

MasterCATZ, 2019-12-29:
yes AU , I will look into EVA boot-loader , thank you

er13, 2020-01-02:
> The files are (usually) present at your own build system already ... look at `tools/make/yourfritz-host/eva_tools` folder.

Minor correction, it's `tools/yf/eva_tools`

closed 2020-01-02 by er13

PeterPawn, 2020-01-03:
Funny ... my first thought was `tools/yourfritz/eva_tools` (https://github.com/Freetz/freetz/issues/257#issuecomment-569407786, 1st version, later edited) and after it wasn't found there with my first attempt (the `yf` abbreviation is clearly possible, but was unknown to me), I've found/composed a wrong path (content below the `make` folder is usually static, as far as I know).