Freetz/freetz #258: Can't flash Freetz on FRITZ!Box 7530

davideakc, 2020-01-05:
Hi,

I've a FRITZ!Box 7530, international, firmware 164.07.12.
I've installed a virtual Linux machine with Freetz-Linux-1.5.6 (Ubuntu-19.10 LTS 64 Bit) on it. I've used [this](https://svn.boxmatrix.info/freetz-ng/trunk/) repository.
I've created my first image file and I've tried to flash it with the FRITZ!Box GUI but it seems impossible because the page says that the file is not compatible with my version of FRITZ!Box (I've choose my version and all the other settings on freetz menu before making the image file).
I searched on internet and most of german forums (I'm italian, I speak only italian and english) say to use EVA FTP.
So I followed a guide and tried to flash my image.
The installation stucks here:

> 220 ADAM2 FTP Server ready
USER adam2
331 Password required for adam2
PASS adam2
230 User adam2 successfully logged in
SYST
215 AVM EVA Version 1.3584 0x0 0x46409
TYPE I
200 Type set to BINARY
MEDIA SDRAM
200 Media set to MEDIA_SDRAM
P@SW
227 Entering Passive Mode (192,168,178,1,12,0)
RETR env
150 Opening BINARY data connection
226 Transfer complete
SETENV memsize 0x0657db00
200 SETENV command successful
SETENV kernel_args_tmp mtdram1=0x8657db00,0x88000000
200 SETENV command successful
TYPE I
200 Type set to BINARY
MEDIA SDRAM
200 Media set to MEDIA_SDRAM
P@SW
227 Entering Passive Mode (192,168,178,1,12,0)
STOR 0x8657db00 0x88000000
150 Opening BINARY data connection

After that, the router starts normally and nothing happens on the console.
Why?
Thanks.

PeterPawn, 2020-01-05:
1. You're using a fork from a different source with an own support, did you realize this?

2. https://www.ip-phone-forum.de/threads/eva_to_memory-failing-for-7530.301400/post-2303077 - if you're using my `eva_to_memory` tool with your 7530 ... the output shown above is clearly from a session, where the memory size was limited to 128 MB (0x08000000).

3. Adding the third parameter to the call as mentioned under the link above, should solve the problem. Otherwise provide some more info please - starting with the command line used to call the script, that produces the output at your comment.

davideakc, 2020-01-05:
Oh, I realize now that I've used another source.
Ok, I've deleted the old one and now I'm using yours.
I've created the file for my FRITZ!Box, now what I have to do?
Because I was using files called "start.sh", "eva to memory.sh" and "eva-login.sh" that in your source I don't find anymore.
How can I upload the image file to my router?
If I type on Linux:

`eva_to_memory myimage.image 192.168.178.1  0`

nothing happens even if I reboot my FRITZ!Box.
Is there a guide on how to use EVA or can you please explain how to do it?

Thanks.

PeterPawn, 2020-01-05:
The needed files may be found in your Freetz build system under "tools/yf/eva_tools".

I've no idea, where you've got the mentioned files "start.sh", "eva to memory.sh" (really with spaces in the name?) or "eva-login.sh" from, but obviously those files were not authored by me and therefore I can't help you anymore, how to use them or how to identify any errors from using.

But using the files from "eva_tools" folder of YourFritz project is explained here: https://www.ip-phone-forum.de/threads/wie-verwende-ich-denn-nun-die-skript-dateien-aus-yourfritz-eva_tools.298591/ - use any (really functioning) online translation tool (e.g. DeepL) to translate it to English or Italian ... at your option.

The used FRITZ!Box device has to be waiting in the bootloader already, if you want to use the "eva_to_memory" script as shown.

davideakc, 2020-01-05:
I've used this command for installing Freetz on my FRITZ!Box 7530:

`./eva_to_memory img.image.in-memory 192.168.178.1 0`

It worked perfectly.

Thanks for your help :)

closed 2020-01-05 by davideakc