Freetz/freetz #265: FRITZ!Box 6591 Cable Support?

soerenkornetzki, 2020-02-19:
Hi, I was wondering whether there will be a firmware for the FRITZ!Box 6591 Cable?
Some time ago, the restriction on ISP issued cable modems was lifted.
I would love to see WPA Enterprise (wlan username password authentication) on the free available cable boxes as well.

f-666, 2020-02-19:
You can get more details about the 6591 here:
[(https://www.ip-phone-forum.de/threads/fb-6591-verschiedenes.303332/)]

The only known way (currently) to get a modified FW onto a box with the latest bootloader is desoldering the flash chip, program it externally and solder it onto the PCB again. Not many people will do that, so it isn't a priority to port Freetz to this box.

f-666, 2020-02-24:
Accoring to fesc an updated bootloader version is able to flash via FTP, so a Freetz port could be possible.