Freetz/freetz #267: build mit net-snmp für die 6590 bricht mit fehlermeldung ab.

Conan179, 2020-02-24:
Hallo erst mal.
Ich wollte für meine 6590 ein freetz-ng image bauen.
Das minimal Image, ohne änderung geht ohne probleme.

Jetzt habe ich net-snmp hinzugefügt.
Jetzt kommt beim build immer diese Fehler Meldung:
freetz@freetz-linux:~/6590a$ make
cmd() { PATH="/home/freetz/6590a/toolchain/build/i686_gcc-4.7.4_uClibc-0.9.33.2-nptl/i686-linux-uclibc/bin:/home/freetz/6590a/toolchain/build/i686_gcc-4.7.4/i686-pc-linux-gnu/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" LD_RUN_PATH="/usr/lib/freetz" FREETZ_LIBRARY_DIR="/usr/lib/freetz" make  "$@"  || { printf "\n\\033[33m%s\\033[m\n" "ERROR: Build failed.";  exit 1; } };        if [ -e source/.echo_item_start -a ! -e source/.echo_item_build ]; then echo -n "building... "; touch source/.echo_item_build; fi; cmd -C source/target-i686_gcc-4.7.4_uClibc-0.9.33.2-nptl/netsnmp-5.7.1 \
        LDFLAGS=" -ldl"
make[1]: Entering directory '/home/freetz/6590a/source/target-i686_gcc-4.7.4_uClibc-0.9.33.2-nptl/netsnmp-5.7.1'
making all in /home/freetz/6590a/source/target-i686_gcc-4.7.4_uClibc-0.9.33.2-nptl/netsnmp-5.7.1/snmplib
make[2]: Entering directory '/home/freetz/6590a/source/target-i686_gcc-4.7.4_uClibc-0.9.33.2-nptl/netsnmp-5.7.1/snmplib'
make[2]: Leaving directory '/home/freetz/6590a/source/target-i686_gcc-4.7.4_uClibc-0.9.33.2-nptl/netsnmp-5.7.1/snmplib'
making all in /home/freetz/6590a/source/target-i686_gcc-4.7.4_uClibc-0.9.33.2-nptl/netsnmp-5.7.1/agent
make[2]: Entering directory '/home/freetz/6590a/source/target-i686_gcc-4.7.4_uClibc-0.9.33.2-nptl/netsnmp-5.7.1/agent'
/bin/bash ../libtool  --mode=link /home/freetz/6590a/toolchain/build/i686_gcc-4.7.4_uClibc-0.9.33.2-nptl/i686-linux-uclibc/bin/i686-linux-uclibc-gcc -DNETSNMP_ENABLE_IPV6 -fno-strict-aliasing -march=atom -mtune=atom -Os -pipe -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Ulinux -Dlinux=linux  -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64  -I/usr/lib/x86_64-linux-gnu/perl/5.28/CORE   -rpath /usr/lib -ldl -version-info 30:1:0 -o libnetsnmpagent.la helpers/all_helpers.lo helpers/baby_steps.lo helpers/bulk_to_next.lo helpers/cache_handler.lo helpers/debug_handler.lo helpers/instance.lo helpers/mode_end_call.lo helpers/multiplexer.lo helpers/null.lo helpers/old_api.lo helpers/read_only.lo helpers/row_merge.lo helpers/scalar.lo helpers/scalar_group.lo helpers/serialize.lo helpers/snmp_get_statistic.lo helpers/stash_cache.lo helpers/stash_to_next.lo helpers/table.lo helpers/table_array.lo helpers/table_container.lo helpers/table_data.lo helpers/table_dataset.lo helpers/table_iterator.lo helpers/table_row.lo helpers/table_tdata.lo helpers/watcher.lo agent_handler.lo agent_index.lo agent_read_config.lo agent_registry.lo agent_sysORTable.lo agent_trap.lo kernel.lo snmp_agent.lo snmp_vars.lo mibgroup/utilities/execute.lo mibgroup/mibII/vacm_conf.lo mibgroup/snmpv3/usmConf.lo  ../snmplib/libnetsnmp.la   -ldl -Wl,-E  -fstack-protector-strong -L/usr/local/lib  -L/usr/lib/x86_64-linux-gnu/perl/5.28/CORE -lperl -ldl -lm -lpthread -lc -lcrypt  -lm
libtool: link: /home/freetz/6590a/toolchain/build/i686_gcc-4.7.4_uClibc-0.9.33.2-nptl/i686-linux-uclibc/bin/i686-linux-uclibc-gcc -shared  -fPIC -DPIC  helpers/.libs/all_helpers.o helpers/.libs/baby_steps.o helpers/.libs/bulk_to_next.o helpers/.libs/cache_handler.o helpers/.libs/debug_handler.o helpers/.libs/instance.o helpers/.libs/mode_end_call.o helpers/.libs/multiplexer.o helpers/.libs/null.o helpers/.libs/old_api.o helpers/.libs/read_only.o helpers/.libs/row_merge.o helpers/.libs/scalar.o helpers/.libs/scalar_group.o helpers/.libs/serialize.o helpers/.libs/snmp_get_statistic.o helpers/.libs/stash_cache.o helpers/.libs/stash_to_next.o helpers/.libs/table.o helpers/.libs/table_array.o helpers/.libs/table_container.o helpers/.libs/table_data.o helpers/.libs/table_dataset.o helpers/.libs/table_iterator.o helpers/.libs/table_row.o helpers/.libs/table_tdata.o helpers/.libs/watcher.o .libs/agent_handler.o .libs/agent_index.o .libs/agent_read_config.o .libs/agent_registry.o .libs/agent_sysORTable.o .libs/agent_trap.o .libs/kernel.o .libs/snmp_agent.o .libs/snmp_vars.o mibgroup/utilities/.libs/execute.o mibgroup/mibII/.libs/vacm_conf.o mibgroup/snmpv3/.libs/usmConf.o   ../snmplib/.libs/libnetsnmp.so -lcrypto -L/usr/local/lib -L/usr/lib/x86_64-linux-gnu/perl/5.28/CORE -lperl -ldl -lpthread -lc -lcrypt -lm  -march=atom -mtune=atom -Os -Wl,-E -fstack-protector-strong   -Wl,-soname -Wl,libnetsnmpagent.so.30 -o .libs/libnetsnmpagent.so.30.0.1
i686-linux-uclibc-gcc: error: unrecognized command line option '-fstack-protector-strong'
make[2]: *** [Makefile:617: libnetsnmpagent.la] Error 1
make[2]: Leaving directory '/home/freetz/6590a/source/target-i686_gcc-4.7.4_uClibc-0.9.33.2-nptl/netsnmp-5.7.1/agent'
make[1]: *** [Makefile:652: subdirs] Error 1
make[1]: Leaving directory '/home/freetz/6590a/source/target-i686_gcc-4.7.4_uClibc-0.9.33.2-nptl/netsnmp-5.7.1'

ERROR: Build failed.
make: *** [make/netsnmp/netsnmp.mk:139: source/target-i686_gcc-4.7.4_uClibc-0.9.33.2-nptl/netsnmp-5.7.1/agent/.libs/snmpd] Error 1

Googlen brachte nichts.

er13, 2020-02-24:
`-fstack-protector-strong` wird erst seit [gcc-4.9.x](https://outflux.net/blog/archives/2014/01/27/fstack-protector-strong/) unterstützt.

Ansonsten fällt auf:

```
[skipped] ../snmplib/libnetsnmp.la -ldl -Wl,-E -fstack-protector-strong -L/usr/local/lib -L/usr/lib/x86_64-linux-gnu/perl/5.28/CORE -lperl -ldl -lm -lpthread -lc -lcrypt -lm
```

6590 ist eine 32-bit Box. Das mit `-L/usr/lib/x86_64-linux-gnu` kann also nicht funktionieren und generell sind die host-Pfade beim Linken fürs target zu vermeiden.

Da Du selber schreibst, dass Du das Problem mit freetz-ng hast, werde ich nichts weiter tun. Es sei denn, Du stellst das Problem auch mit freetz nach und hängst die entsprechende `.config` hier an.

Bis dahin `closed`.

closed 2020-02-24 by er13

Conan179, 2020-02-25:
>Es sei denn, Du stellst das Problem auch mit freetz nach und hängst die entsprechende `.config` hier an.
Diese fehler Meldung bekomme ich NUR im Freetz, im freetz-ng bekomme ich eine andere fehlermeldung, beim selben paket namen.

Ich hab zuerst freetz-ng probiert, mit der verlinkten fehler meldung.

Dann habe ich freetz vom github neu ausgecheckt und dort probiert, zuerst ein image für die 6590 ohne änderungen, einfach nur den Box Typ eingestellt und dann nur net-snmp aus gewählt.
Die .config anzuhängen ist kein problem.
[PUNKTconfig.txt](https://github.com/Freetz/freetz/files/4248906/PUNKTconfig.txt)

Ich möchte nur klar stellen, das ich NICHT den fehler vom freetz-ng hier behoben haben will.
ICh bekomme beim freetz eine andere fehlermeldung.