Freetz/freetz #270: unable to update user interface

rishihashbinary, 2020-03-01:
Hi, I am trying to update the user interface of my fritxbox 6820 v1, I followed the following steps:
1. Download the freetz source code from https://github.com/Freetz/freetz
2. Made changes in the folder freetz/build/modified (also tried changing freetz/build/original )
3. did `make menuconfig`
4. `make`
now I am able to get the * .image file in the images directory, also I am able to flash the router at mtd4 , but unfortunately, my changes are not reflected in the default page displayed at http://192.168.178.1/
Default firmware on the router is 7.01, I can make out the upgrade since the version post flashing is changed to 7.13.

Changes I am trying to make:
I changed the image for logo_fritzDiamond.svg (kept the same file name, but different image)

can someone, please help me in case I a doing anything wrong with the flashing or compilation process? Thanks in advance.

PeterPawn, 2020-03-01:
Script your needed changes as a batch in the `all` function of this shell script: https://github.com/Freetz/freetz/blob/master/fwmod_custom

Afterwards the call to `make` will use `fwmod` to unpack the firmware as needed, modify it (here your own additions in the script above get called) and pack it to a new firmware image.

rishihashbinary, 2020-03-03:
I tried changing image file  logo_fritzDiamond.svg
repacked it using
`./fwmod -p -d unpacked_firmware dl/fw/FRITZ.Box_6820v1_LTE-de-en-es-it-fr-pl-07.13.image`
and used the following command to flash the router
`./EVA-FTP-Client.ps1 -Address 192.168.178.1 -Verbose -Debug -ScriptBlock { BootDeviceFromImage /home/rishi/freetz/images/6820_07.13.en_20200303-140202.image.in-memory }`
but still i cannot see the updated image in my firmware.

How do I know if router was flashed or not?

PeterPawn, 2020-03-03:
You have to apply your own changes under `fakeroot` control, too, or they'll (or at least they may) get ignored (e.g.: https://www.ip-phone-forum.de/threads/cp-failed-to-extend.274703/post-2119734). Why don't you regard the tips, you've got already (https://www.ip-phone-forum.de/threads/customizing-freetz-user-interface.306059/post-2359990)?

And the question, whether a new firmware image was installed successfully, is only the second one ... if you want to be sure it got installed, apply it to a partition with an older version - yes, it's so simple. 

You may use the same workflow to flash this different version (and get experiences, how to install via EVA, at the same time) or you may use the recovery program from manufacturer.

Here are both points a possible source of your problem(s) ... first you have to ensure, whether your changes are incorporated in the new image. You could try to unpack it manually again and check, whether the file in question was exchanged correctly. If it was not, nobody should expect to see a difference in the GUI finally.

If you write a correct file with the shown PS script to the box and it starts this image and begins to blink with the INFO LED, the flash process was started and if this process was finished successfully, the box gets rebooted.

Because the installation in this manner doesn't change the environment variable `linux_fs_start`, there isn't another indication, whether the new image was installed successfully, beside the fact, that the device could be started from this "partition set". 

If it already contains an original image for one version and you're installing the same version again - whether by "accident" (due to a wrong workflow during modification) or by intention, doesn't matter - you're unable to find any difference afterwards.

rishihashbinary, 2020-03-03:
Hi Peter,

> You have to apply your own changes under fakeroot control, too, or they'll (or at least they may) get ignored

I just changed the svg file, which is a picture file for logo, I believe I don't need fakeroot, since there is nothing to test by executing it. Let me know if I missed something in case of understanding the purpose of fakeroot.
 
> Here are both points a possible source of your problem(s) ... first you have to ensure, whether your changes are incorporated in the new image. You could try to unpack it manually again and check, [...]

I extracted files, and I see my updated images in the extracted folder, but cannot find the same changes in the router user interface once I execute BootDeviceFromImage.

If you write a correct file with the shown PS script to the box and it starts this image and begins to blink with the INFO LED, the flash process was started and if this process was finished successfully, the box gets rebooted.

the only LED blinks during the process is marked in this picture -> https://ibb.co/YDL7Mmt
Thanks, I really appreciate your support on this.

PeterPawn, 2020-03-03:
> I extracted files

Show the comparison (e.g. using a SHA-1 check) of the original and the modified (picture) file ... but after really(!) unpacking both (firmware) images (the original and the modified one) - up to the single files in the SquashFS image (and best you show us, too, how you've extracted these structure and here I mean "en detail" and not only as a description like the cited above).

rishihashbinary, 2020-03-03:
I tried generating zip version of the compiled image, in `make menuconfig` I found an option in `Firmware Packing (fwmod) options >> Pack Filesystem into archive` 
this option creates a zipped version which I then uncompress and check for changes.

Checksum for 2 files
```
rishi@rishi-desktop:~$ sha256sum /home/rishi/freetz-git/freetz/dl/fw/FRITZ.Box_6820v1_LTE-de-en-es-it-fr-pl-07.13.image 
f27b913cc7289d8fbe16a7f28bf872d7d4b93a7878f4ad74b4b7a10ea7f3dc2f  /home/rishi/freetz-git/freetz/dl/fw/FRITZ.Box_6820v1_LTE-de-en-es-it-fr-pl-07.13.image

rishi@rishi-desktop:~$ sha256sum /home/rishi/freetz-git/freetz/images/6820_07.13.en_20200303-191846.image
d8e723aa240394a0b7d6d0011758137735fdbf236c953750de3d986f004fac16  /home/rishi/freetz-git/freetz/images/6820_07.13.en_20200303-191846.image
 
```

PeterPawn, 2020-03-03:
And once again ... why don't you even TRY the tips, you've got already?

You're comparing the WHOLE images ... why? It's not really surprising, that they're different. These files are TAR-balls and if only one file date within is different, the checksum (crypto or not) over the whole file gets different.

If you want to verify, that your changes were applied successfully, you have to unpack these firmware image files to SINGLE files (that means, to unpack even the "filesystem_core.squashfs" file from the inner "ext2" image of the "filesystem.image" file from the "firmware image") - or where did you replace your SVG file? In an image? That's something, what _I_ can't believe.

And it doesn't matter, whether YOU may believe it or not, that your problem is (with high possibility) related to `fakeroot` ... simply read (and understand) the man page of `fakeroot` (especially for the `-i` and the `-s` option - and the comments on the latter one) and there's no further need to "believe in anything" ... this should get replaced by sure knowledge.

These "environment files" record inode numbers and file attributes and IF your changes tamper with an inode number (e.g. `rename` or `remove` it, what may even occur while copying a file, depending on the commands used for), this has to be recorded in the used environment file. 

Because the cache file is created during "unpack" and used during "pack", any unrecorded changes in-between MAY get ignored and this has usually only an impact on changes to EXISTING files. Is this "explanation" sufficient, to make you believe?

If it's enough ... if you put your "cp" command (or whatever you want to use here) in the "fwmod_custom" script, as I recommended already two days ago, your command is executed automatically in the `fakeroot` environment and you may ignore further (required) steps, needed to find the correct cache file (it's created with a name, that's based on the MD5 sum of the "original" image in a sub-directory of the base directory for "Freetz"), that you have to specify for an own `fakeroot` call.

rishihashbinary, 2020-03-09:
Just an update on the trail,

1. I made changes in fwmod_custom file's all() function, the script gets called, even did it using `fakeroot` environment 
`${FAKEROOT} -i ${FAKEROOT_CACHE} -s ${FAKEROOT_CACHE} -- cp /home/rishi/freetz-git/freetz/custom/main.css /home/rishi/freetz-git/freetz/build/original/filesystem/usr/www/avm/css/rd/`
and
`${FAKEROOT} -i ${FAKEROOT_CACHE} -s ${FAKEROOT_CACHE} -- cp /home/rishi/freetz-git/freetz/custom/main.css /home/rishi/freetz-git/freetz/build/modified/filesystem/usr/www/avm/css/rd/`
with this I got the following error while executing my custom script
`fakeroot: nested operation not yet supported`
probably because it was already running under fakeroot and i was invoking anther fakeroot instance from within.
2. However removing the fakeroot prefix, my custom script was been called, and I can see an updated file in the generated structure 
though flashing the system through `BootDeviceFromImage` didn't get me the updated web page (as per my changes) in the loaded firmware. 

Thanks for all your help and direction, probably I will give a couple of more tries else will move to another device.