Freetz/freetz #277: make under fresh git clone results in umask error

ziggystar, 2020-03-22:
I'm working under native Linux (Ubuntu) and I get the following error:

    Makefile:101: *** Please set umask to 0022 and then unpack/checkout again.  Schluss.

This is caused by 

```make
# Folder root/ needs 755 permissions
ifneq ($(shell stat -c %a make/mod/files/root),755)
$(error Please set umask to 0022 and then unpack/checkout again)
endif
```
It is desirable to be able to simply checkout and compile. What are you trying to achieve? Is the umask really the problem or the permissions of the files? 

Maybe the problem can be solved with client-side hooks in git.

er13, 2020-03-23:
All the files from the checkout matching the pattern `{make,addons}/*/files/root/...` are copied with their respective access rights directly to the firmware, i.e. "access rights in the checkout" = "access rights in the firmware". This is a design decision made many many years ago.

The alternative would be to specify the firmware access rights in a separate file, but this has been considered as less convenient and thus is not implemented.

So, please set the umask as requested.

closed 2020-03-23 by er13

ziggystar, 2020-03-24:
@er13 Can't the same be achieved with something like the following within the `Makefile` itself?

    find . -path "*/files/root" -exec chmod -R go-w "{}" ";"

This would have the added benefit of assuring that all files have the correct permissions before building the image.

I know this is the same as "specify the firmware access rights in a separate file", where the file is the `Makefile`. But you do not need much code to achieve the same as `umask 022`.

er13, 2020-03-24:
@ziggystar:

This is not about `group & others are not allowed to write`, this is more about `group & others are allowed read & execute/list` (others are more important). Some people use `0077` as umask, resulting in `/` would only be accessible for the root user and nobody else.

To sum it up, your `find .. -exec ..` statement doesn't solve the problem.