Freetz/freetz #278: Unable to execute unsquashfs4-avm-be

KarstenDE, 2020-03-23:
I am unable to execute the `unsquashfs4-avm-be` Command. The file is not executable.

I want to use it to extract the lantiq dsl-firmware and use it in openwrt.
Any hints?

Thanks!

PeterPawn, 2020-03-23:
If this is the only intention (and you know, how to dissect an original firmware image manually), you may try these (pre-built) files, if one for your platform is contained:

https://github.com/PeterPawn/yf_bin/tree/master

KarstenDE, 2020-03-23:
I already downloaded this file `https://github.com/PeterPawn/yf_bin/blob/master/squashfs/i686/unsquashfs4-avm-be`

I tried to use it on my Raspbian x86

But it is not executable :-(

What is required after downloading?

KarstenDE, 2020-03-23:
> and you know, how to dissect an original firmware image manually

If there is another Way to get the dsl-Firmware. I am happy to know it :-)  
Until now I try it with 7zip and this tool here ...

er13, 2020-03-23:
> I tried to use it on my Raspbian x86
> But it is not executable :-(
> What is required after downloading?

`chmod 755`?

PeterPawn, 2020-03-23:
It's not really surprising ... due to the shown path, you're trying to run the x86 version on your Raspberry Pi platform.

And keep this in mind: https://github.com/Freetz/freetz/issues/278#issuecomment-602768404 - but usually a checkout using "git" should set the needed access mode flags. 

If you're downloading the single file, a `chmod` is always needed (or should be needed, because no-one should set a "x" flags from the `umask` settings, if not necessary) and this fact should have been "assimilated" by your "Linux skills" in the past already.

Hopefully you're transferring and storing the binary file in the right manner, because GitHub usage may be a little bit tricky and less-experienced users try often to store something other than the "raw" data - that could be another possible pitfall in your case, leading to the same result/message.

But this is the Freetz repo ... and even if there's a way to generate the needed binary with Freetz, too, it's not really an "issue". 

It's simply an "user failure" for the provided options (to build the tools yourself) and/or some further commands ... you're better of with any additional questions in the "support board" for Freetz: https://www.ip-phone-forum.de/forums/freetz.525/ - as a "cherry on top", this board uses (in most threads) the German language.

If you agree, you should close this issue yourself and move over to the IPPF (link is above).

KarstenDE, 2020-03-25:
Hmm. In General I used Raspbian x86 in a virtual Box.

But anyway I tried multiple ways:
- using the old freetz build environment
- downloaded precompiled (as described above)

With a full clone from the GitHub repo.

And I used `chmod` too.

My next try would be installing a current fedora version.

PeterPawn, 2020-03-25:
The i686 version from my `yf_bin` repo uses ATOM instruction set (it's intended for Intel Puma based devices primarily) and will cease to work on processors (or emulations) without support for the `movbe` machine  instruction (see `flags` from `/proc/cpuinfo`).

But this would lead to an "illegal instruction used" error (while trying to use the binary with an image file, but the '-h' option should still work without further trouble) and not to the reported one ("not executable").

Once again ... you should move on to the IPPF BBS (imho); this was my last reaction/attempt to answer here, sorry (personal decision). 

This feature is for real issues, with a (probable) cause from the provided code of the project, and not really to support users with their "usage problems". Please read this page carefully: https://freetz.github.io/wiki/help/howtos/common/newbie/errors.html

PeterPawn, 2020-05-04:
@er13 o. @f-666 

Kann mal bitte einer von Euch hier zumachen? Das hatte sich dann ins IPPF verlagert, aber auch da irgendwie zu keinem Ergebnis geführt ... aber es ist definitiv keine "Issue" von/für Freetz.

KarstenDE, 2020-05-04:
No issue for GitHub

closed 2020-05-04 by KarstenDE

PeterPawn, 2020-05-04:
Danke.