Freetz/freetz #301: Add support for the current version of Fritz!OS 07.20 for Fritz!Box 7590

e7p, 2020-07-07:
Yesterday, AVM released a major upgrade for Fritz!Box 7590, however mostly updating the kernel from 3.10.104 (07.13) to 4.9.198 (07.20) and many differences in the web interface backend, if I'm correct.

So maybe there is a plan to also support the new 07.2X software in Freetz? I'll try how far I can make use of a modded 07.20 firmware, but most easiest would be if it would directly be supported in Freetz.

PrinzVonBillAir, 2020-07-07:
Freetz-NG has all the labor firmwares (and the new 7590 retail firmware) integrated. However, I don't know if they actually work after the build process is done since I don't use Freetz-NG. So there might be a solution available already?

e7p, 2020-07-07:
Okay, sorry for any inconveniences, I didn't yet hear of Freetz-NG. I could now basically compose a new 7.20 freetz image using freetz-ng, so I would consider this issue closed for the original freetz project. Thanks!

closed 2020-07-07 by e7p