Freetz/freetz #324: 6890 LTE no sim card found

WileC, 2020-10-08:
Aus irgendeinem bislang unbekannten Grund wird seit "freetz-master-f4147ce9f" die eingelegte SIM-Karte nicht mehr gefunden.

Derzeit nutze ich "freetz-master-2b230ce5b"... Den Syslog beim boot habe ich mal angehängt.

[syslog.txt](https://github.com/Freetz/freetz/files/5347165/syslog.txt)

f-666, 2020-10-08:
Wenn man sich die Änderungen von f4147ce9f zu 2b230ce5b anschaut, ist nichts dabei, was offensichtlich diesen Fehler verursachen würde. Auch im syslog.txt ist im Hinblick auf das Modem nichts offensichtlich falsch (bei [65.740000]).

Kannst du wirklich den Fehler provozieren, wenn du zwischen den beiden Freetz Revisionen umschaltest?

WileC, 2020-10-08:
Das habe ich so noch nicht probiert, da diese Version nicht mehr auf der Box ist. Mit einer neu compilierten “git checkout b4...” ging’s dann heute auch nicht mehr. 

Ich werde das mal auf ner 6820 mit aktuellem Freetz Master und der SIM-Karte aus der 6890 Testen. Klappt’s da auch nicht.. liegt’s entweder an der Karte oder der Software. Als Vergleich dient dann natürlich die originale Firmware.

WileC, 2020-10-27:
Ich hab die "alte" SIM-Karte mal an meiner 6820 mit dem gleichen freetz-master getestet und siehe da, sie funktionierte nicht. Nun habe ich eine neue SIM-Karte, welche in der 6820 funktioniert. Wenn sie demächst dann auch in der 6890 funktioniert, handelte es sich offensichtlich um einen Defekt der SIM-Karte...

WileC, 2020-11-06:
So, ich habe nun eine neue SIM-Karte eingesetzt und diese funktioniert sowohl in der 6820 als auch in der 6890. Also wurde das Hauptproblem wohl durch eine defekte SIM-Karte ausgelöst...

WileC, 2020-11-12:
Also mir hats jetzt die neue SIM-Karte wieder durchgesemmelt. "SIM-Karte nicht gefunden" obwohl diese eingelegt ist... Habe bereits an AVM getwittert, ob das vielleicht ein Hardware-Problem sein könnte? .. Bislang hatte ich wieder mit Handys noch mit der 6820 ein SIM-Karten-Problem...

WileC, 2020-11-16:
Antwort vom AVM-Support:

> Es kam immer wieder vereinzelt zu Fällen mit nicht mehr funktionsfähiger SIM-Karte nach einiger Laufzeit. Die SIM-Karten lassen sich zunächst über mehrere Wochen oder Monate ohne Probleme in der FRITZ!Box 6820 LTE bzw. FRITZ!Box 6890 LTE nutzen, fallen aber nach einiger Zeit aus.

Wir haben das Problem mittlerweile soweit eingrenzen können, dass ausschließlich bestimmte SIM-Karten-Typen betroffen sind. Bei näherer weiterer Betrachtung konnte die Ursache für den Ausfall der SIM-Karten dahingehend eingegrenzt werden, dass dieser ausschließlich im Zusammenhang mit bestimmten lesenden Zugriffen auf die SIM-Karte auftritt. SIM-Karten beinhalten einen eigenen Prozessor, ein eigenes Betriebssystem und eigene vom Hersteller bzw. Netzanbieter zu verantwortende Applikationen, die für das Mobilfunkgerät selbst weder zugreifbar noch sichtbar sind. Wir verorten auf Basis der momentanen Erkenntnislage einen Zusammenhang mit genau dieser SIM-Applikationsebene. Einige dieser SIM-Kartentypen haben sich schon vor geraumer Zeit im Kontext mit anderen Mobilfunkgeräten als ausfallanfällig erwiesen und wurden von einzelnen Anbietern daher teils aus dem Verkehr gezogen.

FRITZ!OS 7.21 enthält einen Lösungsansatz, der die latent zur Selbstzerstörung neigenden SIM-Kartentypen vor einer solchen künftig schützen sollte. Bitte prüfen Sie, ob diese Firmware bei Ihnen den erwarteten positiven Effekt hat und teilen Sie uns bitte nach entsprechendem Einsatz Ihr Feedback mit. Sollte unsere Firmware das Problem bei Ihnen nicht lösen, sehen wir vorerst als sinnvollsten Ansatz an, die Verwendung einer SIM-Karte aus den betreffenden Chargen zu vermeiden.

Bitte wenden Sie sich daher an Ihren Anbieter mit der Bitte, die bei Ihnen vorliegenden SIM-Karten aus diesen Chargen aus dem Verkehr zu ziehen und Ihnen im Austausch SIM-Karten neueren Produktionsdatums bzw. SIM-Karten aus einer nicht von dem Problem betroffenen Charge bereitzustellen.

SIM-Karten mit folgenden aufgedruckten Kartenkennungen sind von dem Fehlerbild potentiell betroffen:

    C/WE
    C/W7
    C/W2
    V3
    Vu 0503

Die betreffende Buchstaben/Zahlen-Kennung befindet sich je nach Kennung/SIM-Kartentyp jeweils an etwas anderer Stelle, aber immer auf der Rückseite der eigentlichen Chipfläche und dort meist rechts neben dem unteren Teil der aufgedruckten SIM-Kartennummer (IMSI) (z.B. bei den Kennungen C/Wx) bzw. in der kompletten unteren Zeile (z.B. bei Kennung Vu xxxx). Wir können nicht ausschließen, dass SIM-Karten mit weiteren Kennungen von diesem Problem ebenso betroffen sind. Bisher haben wir aber im Zuge unserer Untersuchungen keinerlei SIM-Karten mit abweichenden Kennungen als anfällig identifizieren können.

closed 2020-11-16 by WileC