Freetz/freetz #344: Upload of own image failes on 7412 FritzBox

blub4747, 2021-02-04:
Hello community,

I built myself my own personal image for a FritzBox 7412 with the Freetz built machine on Xubuntu 16.04.
However when I try to upload it to my 7412 

> FRITZ.Box_7412-06.86.image     

I get a message telling me that image is not compatible. 

Which steps am I missing to successfully upload my own image ?

Thx,
blub4747

PeterPawn, 2021-02-04:
Installation of first own image has to be done via FTP session from bootloader - later images (if signed) may be uploaded using AVM's GUI (Freetz' own update mechanism is somewhat outdated).

Explanations, how to install firmware using EVA (the bootloader by AVM), may be found at IP-Phone-Forum (ip-phone-forum.de), the usual "support site" for Freetz. There are threads in english, too - and the process is described more than once (sort your results from search engines by date in descending order to get the newest first and don't fool/confuse yourself by very old ones), so it should not be necessary to write an own question - as long as there's not a (concrete) problem while following the guides/stories.

PeterPawn, 2021-02-04:
And by the way ... this here is not really an issue - it's a "handling error" or "lack of knowledge" and should get closed asap, together with the "knowledge gap".

closed 2021-02-05 by f-666

blub4747, 2021-02-06:
Follow up question on the part about signed images.

Is there within the freetz built machine a option to create signed images ?

PeterPawn, 2021-02-06:
Did you try to search for it? May you describe, what the problem was to find it? It's important to understand such difficulties to offer better experiences for later beginners. 

And to answer the question: Yes, there's an option to sign images: https://github.com/Freetz/freetz/blob/master/config/ui/fwmod.in#L49 and there's an explanation (in the help text from old Trac Wiki) of the signing process and prerequisites, too. It may be available in German only, but there's an offer for sure to translate it (automatically by an internet based service). I was able to find the right page using Google for searching, simply with the keywords `freetz sign images`.